descampsa/ardyno #5: AX12a won't respond after changing id

ThomasSchaschi, 2017-01-26:
Hi there!
Recently I've been playing around with my Dynamixel AX-12as which have worked fine until I executed
the following command: 

`device.write(0x03, 0x05);`

which should change the id of the connected dynamixel to 5

However, after doing this, the motor won't respond to anything at all.
I would be glad if someone could help me out. 
Thanks.

descampsa, 2017-01-26:
Hey,
Have you tried to change the id passed to the DynamixelMotor constructor to 5 after that?

ThomasSchaschi, 2017-01-26:
Hey, 
First of all, thank you for the quick reply ;)

I am just using the example code where the led state changes after the certain duration.
- And yes I have tried that

I am now attempting to use another library to reset the motors.
I

descampsa, 2017-01-26:
Another thing to check, is how you initialize the DynamixelInterface.

If you connect the rx/tx pins directly to the motor, you must use
`HardwareDynamixelInterface interface(Serial);`
If you use a 3-state buffer with a direction pin, you must use
`HardwareDynamixelInterface interface(Serial, 2);` (if 2 is your direction pin)

Since not all of the examples use the same configuration by default, that might be confusing (i also have to fix that)

ThomasSchaschi, 2017-01-26:
Yeah thank you I have noticed that (I've gone through all the C++ files)
That is not the problem since I am able to communicate with them (turn, led etc) but not set the ID.

- So I can do everything but setting a new id

descampsa, 2017-01-26:
Hmm, i have one (untested) idea of what have gone wrong, that's a little bit subtle, so bear with me:

- the write function is template, so you can write different value type easily (mostly uint8_t and uint16_t)
- apparently, an literal value (like 0x05) is considered as a 16bit value by the compiler (i have tested that)
- that means that `device.write(0x03, 0x05);` actually write a 16bit value, so it affect both the 0x03 and the 0x04 registers
- the 0x04 register is the baud rate register, so the baud rate has been modified, that explain why you can't communicate anymore with the servo

If i am not mistaken, the baud rate register has been set to 0x00, that would mean a 2000000 baud rate. 
Changing the baud rate to this value in `interface.begin` should thus make you able to communicate with it again.

ThomasSchaschi, 2017-01-27:
Ok I gave it a shot, but it didn't make a difference. 
Anyway, could you please tell me the command to set the id properly. 
I'm assuming I would have to set it with a
uint8_t
don't I?

descampsa, 2017-01-27:
I have just made a test, and i confirm that i can reproduce your exact problem: by calling `device.write(0x03, 0x05);`, the baud rate has been changed to 2Mbaud.
Note that this baudrate is technically out of spec for the AX-12 (max 1Mbaud according to the datasheet), and the connection might be less reliable if you are in a noisy EM environnment, or use a long cable.

For your question, yes, in the current interface, you must use a uint8_t variable, or specify the template argument explicitly (`device.write<uint8_t>(0x03, 0x05);`) to write correctly a register.
I agree that this is too confusing, i will have to change that also. 

For the specific task of changing the id, i have just pushed a commit that add a `changeID` function, so you just have to use that (this is not yet in the Arduino Library Manger version, only in the git repository).

ThomasSchaschi, 2017-01-27:
Alright, so I guess you made a test.
So I am now gonna download the newer version and give it another shot on the motors I am still able 
to communicate with. 
Do you have any idea how to recover the other servos (I don't want to invest in an Usb2Dynamixel if not necessary)?

descampsa, 2017-01-27:
Have you tried the led example, with a 2Mbaud baud rate on your 'dead' servos?
Be sure to set the id to BROADCAST_ID, and to correctly configure the DynamixelInterface depending of your configuration.
According to my tests, and if there is no other problem, it should work.

descampsa, 2017-02-08:
Hey, have you finally been able to solve the problem?
If not, a solution may be to reset them using their bootloader.
I may indicate you a way to that with an arduino.

ThomasSchaschi, 2017-02-13:
I am really sorry that I have not replied for a long time since I have not had the opportunity to test the code on the servos. What it resulted in: The setting of the id now works, however I cannot establish communication with the 2 servos I messed up. 
I'm still trying to recover them. 
Thanks for the support.

descampsa, 2017-02-13:
Hey, no problem, glad to hear that it works now with the updated code.

For the servos you can't recover, you may try the arduino code in attachment below. 
It will give you access to the motor bootloader, from which you can force a reset of the motor parameters.

Basically, it continuously send "#" characters at 57600 baud, which force the motor to boot in bootloader mode, and then act as a proxy between the arduino usb serial connection and the motor uart.

Do the following:
- first upload the code to the arduino, and, with the arduino connected to the usb port, open the arduino serial console at 57600 baud on your pc.
- connect one (and only one) motor to the arduino, with the data wire connected to the pins 2 and 3. It is important that the data wire must be connected and active before the motor boots. One way to do that is to connect the three motor wires (power wires to gnd/vin) with the arduino only connected  to usb, and then plug a 12v power supply to the arduino.
- If it work, you should see some data on the serial console, and the motor led should stay on
- In the serial console, enter 'r' and press enter. That should force the motor to reset, the led should blink a few times and then stay off. Don't enter anything else or you could mess with the firmware, and then need a Usb2Dynamixel to recover it.
- The motor should now have an ID of 1, and a baud rate of 1Mbaud (and all other parameters reset to their default value).

[dynamixel_bootloader.txt](https://github.com/descampsa/ardyno/files/772318/dynamixel_bootloader.txt)

ihabrazi, 2017-04-21:
Dear Sir :Thank you for your answer and Sorry for the delay I was sick

When I use it 

1) DynamixelInterface &interface=*createSerialInterface(Serial); : It means I was busy TX0/RX0 for arduino méga

But how can I work with TX1/RX1,TX2/RX2, TX3/RX3 . at the same time

2)  can You  send me a programming code for arduino

I'll be glad for that, sir

I want this because I have a different set of servomotors, I can not contro

Maik93, 2018-10-16:
Hi, I'm sorry to open again a closed issue, but I don't know how to contact you directly.

I don't know what's append but today two of my 18 Dynamixels (I'm building an hexapod with 3 motors per leg) stopped working: their red led is permanently on, don't respond to any command and are not listed anymore with any ID (at 1Mbps).

I interface with them with a 74LS241 dual buffer; I don't have an Usb2Dynamixel and it would very impractical for me to obtain one of them. Do you think that it's possible to flash the official firmware following your last post procedure? In that case, have you got the official firmware .bin file? I can't find it neither on the internet nor in any directory of RoboPlus (the official Dynamixel program).

One different way could be interfacing with RoboPlus with an Arduino and the buffer (instead of the Usb2Dynamixel) and use the official program, what do you think about it? In that case I suppose that the Arduino should simply forward the serial traffic to the buffer, similarly to the code of your previous post, isn't it?

Thanks if you'd like to help, I really don't know what to do.

shettysameeksha, 2019-01-19:
if I am unaware of the id set how could I fetch the present ID of AX12A?
the default id of ax12a is changed?

closed 2019-01-19 by ThomasSchaschi

Mecb1407, 2021-05-01:
Hi @Maik93 , You were able to fix your mistake? 

Regards.