descampsa/ardyno #7: Does not work on Arduino due SoftwareSerial.h not found

pradyum, 2017-03-19:
hi , is it necessary to include SoftwareSerial.h , even though i am using a hardware serial port on Arduino due . 

Is there any solution

descampsa, 2017-03-19:
The library should compile fine with SoftwareSerial.h, since it is a standard arduino library.
It will only work on reasonnably recent versions of the IDE, though. 
What version are you using?

pradyum, 2017-03-19:
i use arduino 1.8.1 , and i found that arduino due does not have a SoftwareSerial.h library since it has arm m3 32 bit chip .

descampsa, 2017-03-19:
Ok, then i will add a preprocessor option to disable SoftwareSerial, and disable it by default on due.
For now, you could comment the relevant code manually in the library, it is in DynamixelInterfaceArduinoImpl.h/cpp , in the Arduino/libraries/ardyno/src directory (make sure to update the library to the v0.7 version that i just released).

pradyum, 2017-03-19:
Now i have commented SoftwareSerial.h and other related functions in
DynamixelInterfaceArduinoImpl.h/cpp . But there is another problem , functions setReadMode and setWriteMode are only declared for AVR and Arduino 101 , there is no setReadMode and setWriteMode  for ARM

descampsa, 2017-03-19:
Hmm, yeah, indeed, i have not ported them to arm.
The solution depends of how you connect the motors : if you use tristate buffers with the direction pin, as recommended by robotis, those function are not really needed, and you can just define them empty.

If you connect the rx/tx directly to the motor communication pin, you really need those functions, and need to implement them for arm (which means going into the uart detail in the uc datasheet). If you are interested to implement them, let me know, i will help you the best i can, but i do not have any hardware to test it myself.

pradyum, 2017-03-19:
Thanks it compiled , I am using a tri-state buffer and commented the call functions lines for setReadMode and setWriteMode  in DynamixelInterfaceArduinoImpl.h/cpp.

closed 2017-03-19 by descampsa