descampsa/ardyno #8: getting started troubleshooting - led example not working

eecharlie, 2017-03-23:
I'm in a bit of a hello world situation, using your library to use Dynamixel servos for the first time. 

I have an Arduino UNO hooked up to an AX-12W per your wiring diagram, and I've experimented with both the speed example and the led example. 

Neither is producing any activity from the motor. I modified the speed example to use broadcast address 254, and I see that the led example is already set to write to the broadcast address.

Troubleshooting: The AX12-W led blinks once on power-up as it should. Shorting rx and tx per your wiring diagram prevents serial comm with the host PC, so I connect it only after downloading the program. Is this really correct? 
I can confirm with an oscilliscope that there are is a periodic burst of activity on the data line, which is otherwise at 5v. 
I don't believe any settings, such as bus ID, have been changed on these servos, but if there is some factory memory reset function that would nice to try.

Any suggestions? It would be nice to include a complete and clear hello-world example in your repo, and I think the LED example is nearly that but it's not working for me. 

Thanks!

descampsa, 2017-03-27:
If you have tried the speed example, it is possible that you have changed the communication baudrate of the servo. 
Can you try with 9600 or 57600 as baudrate, instead of 1000000? 
Also, since you use the hardware serial with no buffer, make sure that you create the interface exactly like this:
`HardwareDynamixelInterface interface(Serial);`

And yes, the examples are clearly not as good as they should be, i will fix that as soon as i find some time.

It is normal that you have to disconnect the comm to the PC, since there is only one hardware uart on the Uno, the only way to communicate with the motors and the PC simultaneously is to use a software serial, but then you are seriously limited in baudrate.

eecharlie, 2017-04-18:
Thanks, and sorry for the delay. 

I've just tried using your updated led example. FYI, the const baudrate is not passed as a parameter to interface.begin().  I am using the hardware serial without buffer option:
`// Use this for hardware serial without tristate buffer
HardwareDynamixelInterface interface(Serial);`

It's still not working. I've tried baudrates of 9600, 57600, and 1000000. I'm using a Sparkfun Redboard, with a 9V AC adapter supply, and I can see that the red power LEDs on the Dynamixels (I have 2 chained) are lighting up when they first turn on. 

One odd thing I notice: each time the example code writes to the bus, there is an activity burst on the Arduino RX LED, but *only* if the USB cable is plugged in. It does not happen when running only from AC adapter power.

descampsa, 2017-04-19:
>  FYI, the const baudrate is not passed as a parameter to interface.begin().

oops, yes, you are right, thanks for reporting that.

> there is an activity burst on the Arduino RX LED, but only if the USB cable is plugged in. It does not happen when running only from AC adapter power.

The RX/TX led are controlled by the usb controller, so it is normal that they don't blink when the usb cable is not plugged in. 
It is really the RX LED that blink? It should be the TX LED (at least on the Uno)

> It's still not working. I've tried baudrates of 9600, 57600, and 1000000. I'm using a Sparkfun Redboard, with a 9V AC adapter supply, and I can see that the red power LEDs on the Dynamixels (I have 2 chained) are lighting up when they first turn on.

Ok, not sure what the problem is, then.
The most likely problem is a different baudrate between the motor and arduino, since you have used the old test_speed example. If you remember using different baudrates that 9600, 57600 and 1000000, try those.

Also, 9V is on the low side of the voltage range, and with the reverse protection diode in the Arduino, the actual motor voltage is probably below 8.5V. I doubt this is really the problem, since the motor LED blink once when you connect them, but if you have a 12V power supply, could you try it to be sure?

To be sure, you can also check your oscilloscope results (if you have a dso with uart decode), it should be :
0xff 0xff 0xfe 0x04 0x03 0x19 0x01 0xe0
and 
0xff 0xff 0xfe 0x04 0x03 0x19 0x00 0xe1
alternatively.

If nothing else  work, there is a way to reset the motor to their original config with the bootloader, it is described here : https://github.com/descampsa/ardyno/issues/5#issuecomment-279511638 
Ask me if there is anything unclear with the process.
After that, the motors ID will be reset to 1, but you can change them back with the changeId function.

eecharlie, 2017-04-19:
Thank you! The bootloader reset process worked, and now the LED example is working for me. Here is a fairly clean .ino file if you want to add it as an example/utility to your repo.

[arduino-dynamixel-recovery.ino.txt](https://github.com/descampsa/ardyno/files/937960/arduino-dynamixel-recovery.ino.txt)

eecharlie, 2017-04-20:
(Marking as closed)

closed 2017-04-20 by eecharlie