descampsa/ardyno #9: Controlling different types of servomotors with arduino méga

ihabrazi, 2017-04-21:
Hi sir Adrien Descamps
When I use the following function .  
                  DynamixelInterface &interface=*createSerialInterface(Serial);
It means I was busy TX0/RX0 for arduino méga, but I want to control four different types of servomotors
How can I open ports? TX1,2,3 and RX1,2,3 . At the same time
If you do not mind send me the code

descampsa, 2017-04-21:
Hey,

First, i recommend that you update the library to the 0.8 version, using the arduino library manager.
The new way to create an interface is now :
`HardwareDynamixelInterface interface(Serial);`

You may create several interfaces and attach motors to any of them. 
For example, if you have servo1, with id 1, and  servo2, with id 2, on the first interface (RX1, TX1), and servo 3, with id 1 on the second interface(RX2, TX2), you can do :

`HardwareDynamixelInterface interface1(Serial1);
HardwareDynamixelInterface interface2(Serial2);
DynamixelMotor servo1(interface1, 1);
DynamixelMotor servo2(interface1, 2);
DynamixelMotor servo3(interface2, 1);`

Don't forget to call init with the correct baudrate for each interface, and then you can access each servo normally in the code.

If you want to put several motors on the same interface, you will have to change their ID, because they all have the default ID 1 when you buy them. For that, there is a `changeId` function. Connect only one motor, and do something like:
`DynamixelMotor servo(interface, 1);
servo.changeId(newId);`

The motor will then have the newId ID. Repeat for each motor to assign an unique ID to each.

I see in you message that you have AX-18, MX-12A, RX-24F and XL-320 models. 

For RX-24F, you can't connect them dirrectly to the arduino mega, because they use a RS485 signal, you must use something like a MAX485.

For XL-320, be carrefull that its voltage range is lower than for the other models, so you can't connect them to the same power supply than the other models. Also, it seems that the registers are not exactly the same than for the other models, so some functions might not work as expected.

nitinsonker78, 2020-09-05:
hey i m having very much troboul with converting joint mode to wheel mode in Dynamixel XL-320 with arduino UNo or mega. Please can u help me step by step .
or if the code available for arduino uno please send me.