descampsa/ardyno #13: Improving library usability and robustness

eecharlie, 2017-05-17:
I've found it a bit confusing to jump in and use the library, partly navigating the different classes to find the relevant one, but mostly because there are very few convenience functions and what they do is a bit limited. Suggestions (that I am happy to begin implementing if we are on the same page):

-More getter/setter functions for basic tasks, like changing drive parameters, avoiding normal need to use read() and write()
-All getters & setters are clearly named as such (e.g. setGoalPosition and getCurrentPosition)
-All getters and setters return a boolean that's true if communication was successful
-Getters pass in an argument by reference that they write the requested value to

descampsa, 2017-05-18:
Yes, i mostly agree with you that the convenience functions do not cover everything that they should, and that they lack consistency. 
I just don't want it to become a list of getter/setter for each register, there would be very little added value compared to the `read`/`write` function, but i don't think it is your intention.

So ok for your two first points and the fourth point. For the third point, i would prefer the return value to be the complete status byte, even if it means slightly more typing to check the result.

So that we do not do the same work twice, a propose that i work on the existing functions (in a separate branch for now, since it can break existing code), and let you propose/implement new functions that you think would be usefull.

eecharlie, 2017-05-18:
Sounds good. Reviewing the status codes, they had the forethought to make
STATUS_OK =0 so one can just check for nonzero returns which is just as
clean as a Boolean.

You'll see my pull request is consistent with this except for the return
code type. I can fix this and resubmit.

 I don't have time to implement everything that should be done, I will just
be doing the functions I need for my application :)

Incidentally any tips on tuning feedback control parameters? I have a very
simple 1dof mechanical load and the servo bounces a bit before stopping
which is worse than a $10 non digital servo!

descampsa, 2017-05-18:
> You'll see my pull request is consistent with this except for the return
code type. I can fix this and resubmit.

Yes, it looks good, please fix the return type.

>  I don't have time to implement everything that should be done, I will just
be doing the functions I need for my application :)

Sure, it is already good ;-)

> Incidentally any tips on tuning feedback control parameters? I have a very
simple 1dof mechanical load and the servo bounces a bit before stopping
which is worse than a $10 non digital servo!

Yeah, the control algorithm of the AX-12 is not so great in that regard. 
It usually help to increase the compliance slope, probably to the max value (128) if you want no bounce at all.
You may also increase the compliance margin, but then you lose position accuracy.

descampsa, 2017-05-20:
Regarding the compliance slope, it seems that the documentation is wrong about it.
The documentation says that the slope has only 7 steps, and that, for examples 254 is interpreted the same as 128, but from my tests, 254 clearly results in a higher compliance slope than 128.
On the other side, 255 is probably a 'special' value, because result in a very small slope.

eecharlie, 2017-05-22:
That sort of information would be great in a wiki attached to this repo. As I've gotten to know these servos better (the AX-12A anyway) the documentation is pretty spotty and sloppy. 

Still waiting on anything from me on the pull request?

closed 2018-10-16 by eecharlie