descampsa/ardyno #16: Mega board cannot communicate with Dynamixel MX28T

Lazy12316, 2017-05-31:
Hello Sir,  
            I am new to the Dynamixel Motors. I followed the instructions as You have given in documentation. I uploaded the joint mode code to my Arduino mega, but nothing happens. **Only led of the dynamixel motor** **blinks**. I went through the code, and realized that there is no communication between the motor and the board. Can you please help me know where i am going wrong and the problem. I have set the baudrate and id too.

eecharlie, 2017-05-31:
When you say that only the LED blinks, do you mean that it blinks when the servo powers up, not as a result of commanding it to blink in software? 

Is the servo brand new? How do you have confidence that the baudrate and ID you've set are consistent with the settings in the motor?

descampsa, 2017-05-31:
Did you modify the example code, to use another Serial port for example? If yes, please join the exact code you are using.

How do you connect the motor to the arduino exactly?

Lazy12316, 2017-06-01:
@eecharlie The LED of the Dynamixel Servo blinks normally when powered on. And after uploading the Blink_Led example, the LED blinks Normally. I am Confident about the baudrate and ID of the dynamixel as i modified it using ROBOplus software using a usb2dynamixel connector. I am unable to run the joint_mode example. I went through the code to findout where i am actually going wrong. I found out in Void Setup, there is no communication between the Motor and mega board.

void setup()
{ 
  interface.begin(baudrate);
  delay(500);
  
  // check if we can communicate with the motor 
  // if not, we turn the led on and stop here
  uint8_t status=motor.init();
 if(status!=DYN_STATUS_OK)   **<------------------------ I feel this is, where the problem is**
 {
   pinMode(LED_BUILTIN, OUTPUT);
   digitalWrite(LED_BUILTIN, HIGH);
    while(1);
  }

  motor.enableTorque();  

  // set to joint mode, with a 180° angle range
  // see robotis doc to compute angle values
  motor.jointMode(204, 820);
  motor.speed(speed);
} 

The Dynamixel Servo is not new. 

@descampsa I haven't Modified the code, except for  Dynamixel Id and baudrate.
Here is a Picture of harware connection of the Board and motor.
[image]
**Does it blink continously wathever you do on the arduino? That would mean that the motor has detected an internal error (out of range voltage, overheat, ...)**
I am no sure about this problem.  
Please let me know if the problem persists due to my servo motor or there is any other problem.

descampsa, 2017-06-01:
If i see correctly on your image, you are using the arduino 5V output to power the motor.
You can't do that, the MX28T requires at least 10V.

You can power the arduino with a 12V power supply and take power from the Vin pin (beware to the max current rating of the arduino protection diode, if you use many motors/important loads), or connect the motor power pins directly to the 12V power supply (don't forget to connect the arduino GND in this case)