descampsa/ardyno #19: Can't communicate with RX-28 after changing baudrate

arriras, 2018-06-21:
Hello,
I recently got ahold of a Dynamixel RX-28 actuator/servo, I managed to make the led blink and move the arms through Arduino Nano hardware serial. But I wanted to communicate through USART while moving the servo, so I tried to lower the baudrate of RX-28 to 7534, which seemed to be the lowest possible bitrate, so I could use software serial. Afterwards I couldn't get contact with the servo anymore, I changed the baudrate in the code also to 7534. I tried to reset RX-28, but nothing I did worked anymore.
If anyone has any ideas or solutions, they would be appreciated.
Thanks,
A

closed 2018-06-21 by arriras

descampsa, 2018-06-21:
Thanks for explaining how you fixed your problem.
FYI, there is an example of how to reset a motor in this thread (only tested on AX-12):
https://github.com/descampsa/ardyno/issues/5#issuecomment-279511638

arriras, 2018-06-21:
I always try to explain the solution if I find it, it's just good manners in my opinion.
I tried the motor resetting code you mentioned, but it didn't work.