descampsa/ardyno #24: cannot upload the program which following the same wire connection

Zero23, 2019-10-04:
Hi descampsa,

I am trying the dynamixel wheel mode example code using Arduino Mega 2560. We found that if we connect our motor to serial 0 (TX0 and RX0, wich are pin 0 and 1), the Arduino code cannot be uploaded successfully (with the error message: avrdude: stk500v2_command(): unknown status 0x01  avrdude: stk500v2_disable(): failed to leave programming mode). Then, if we connect to serial 2,3, or 4 (TX1&RX1, TX2&RX2, TX3&RX3), the code can be uploaded successfully, but the motor does not run. So we tried to upload the program with connecting the motor to serial 2, 3, or 4, and after uploading, we switch the wires to serial 1, then the motor works. So I want to ask that how can we enable the same serial with which the program can be uploaded successfully and the motor can work. Besides, since I want to connect multiple motors, how can I make all of the serials work?

Thanks a lot!

descampsa, 2019-10-07:
Hi Zero23,

It is normal that you can't program the arduino with the motor directly connected to the TX0 and RX0 pins, since the serial0 is shared with the usb connection.
If you want to use RX1/TX1 for example, you must change the line 
`HardwareDynamixelInterface interface(Serial);`
to
HardwareDynamixelInterface interface(Serial1);

You can connect as many motors as you want (only limited by the number of ids) on the same serial.

Zero23, 2019-10-14:
Hi descampsa,

Thank you for replying. 
It work for me now. Thank you so much.

closed 2019-10-15 by descampsa